srobo/competition-simulator #88: Change material paths to work on Linux

antoinepetty, 2020-06-01:
[image]
As highlighted by a team on the forum, the paths to certain textures are Windows-specific. They should work on any platform

TomWheal, 2020-06-01:
This is also present on Windows, when installed for a single user not an entire machine. 
This could be what students would do on school computers as it doesn't require admin rights.

PeterJCLaw, 2020-06-01:
Given that it outputs:
> A resource file can be defined relatively to the worlds directory of the current project, relatively to the worlds directory of the default project, relatively to its protos directory (if defined in a PROTO), or absolutely.

It's not obvious to me that any of those options are really what we want. From the looks of it we're trying to use textures from the Webots default projects.

Workarounds therefore include:
- duplicate those textures into our repo or distribution (has licensing questions)
- come up with our own textures
- drop the textures
- something nasty in our controller code which copies them from Webots at runtime

I've not looked into ways to re-use the textures from the default project, though I'm guessing from that error that it's not really a supported case (which seems a shame).

RealOrangeOne, 2020-06-01:
> relatively to the worlds directory of the default project

Looking at the URL in the image, this is actually what we want. In theory just `./textures/metal.jpg` should work.

I'll try and give this a test tomorrow, but if someone does before even better!

antoinepetty, 2020-06-02:
I asked on the discord and it seems that you can't create relative paths to the default projects. We'll have to decide to go with one of Peter's options
https://github.com/cyberbotics/webots/issues/1638

antoinepetty, 2020-06-02:
> > relatively to the worlds directory of the default project
> 
> Looking at the URL in the image, this is actually what we want. In theory just `./textures/metal.jpg` should work.
> 
> I'll try and give this a test tomorrow, but if someone does before even better!

Forgot to mention that I did try this yesterday and it was unsuccessful

PhilR54, 2020-06-12:
Is this still an issue? I can just remove the metal effect from those parts of the robot - it'll just appear with a matt finish which isn't a whole lot different.

antoinepetty, 2020-06-14:
Should have been auto closed

closed 2020-06-14 by antoinepetty