srobo/competition-simulator #93: Add drifting functionality

antoinepetty, 2020-06-01:
Currently use_brake is on by default and no changeable. Would be good to allow teams to have their bot come to a rolling stop https://studentrobotics.org/docs/programming/sr/motors/#stopping-the-motors

antoinepetty, 2020-06-01:
https://github.com/srobo/competition-simulator/blob/master/controllers/sr_controller/sr/robot/motor.py#L77

RealOrangeOne, 2020-06-01:
Simply restoring that code won't quite work, as we need to actually implement the braking.

I believe in the actual SR kit this is done in hardware.  Looks like webots has [braking](https://www.cyberbotics.com/doc/reference/motor#wb_motor_get_brake) which can be used to set the damping force, which would work to ease the brakes. If we set it to something low enough, it'll be as if there's no braking. Our robot doesn't have this braking node attached though, so I guess hard braking is the default with webots.

trickeydan, 2020-06-01:
> I believe in the actual SR kit this is done in hardware.

In the actual kit, "braking" simply shorts the two sides of the motor together, causing an inductive braking effect.

PeterJCLaw, 2020-06-01:
> > I believe in the actual SR kit this is done in hardware.
> 
> In the actual kit, "braking" simply shorts the two sides of the motor together, causing an inductive braking effect.

Yup, pretty sure that counts as "hardware" :joy:

trickeydan, 2020-06-01:
> Yup, pretty sure that counts as "hardware" joy

I know, but I think it's always good to spread information and knowledge :)

PeterJCLaw, 2020-07-04:
I was having a look in the Webot docs and I think that we could approach this using the [force/torque motor controls](https://www.cyberbotics.com/doc/reference/motor#force-and-torque-control) rather than the velocity controls.