srobo/competition-simulator #101: time.sleep uses real time instead of simulation time

antoinepetty, 2020-06-08:
time.sleep uses real time instead of simulation time. These are not the same, and as the simulator is being used, we should be using simulation time.

Quoted from https://studentrobotics.org/forum/viewtopic.php?f=10&t=89&p=340#p340

PeterJCLaw, 2020-06-08:
Nasty, nasty thought -- we could patch `sys.path` so that `import time` returns something which _looks_ like the built-in `time` module, but isn't.

I really don't think that's a good idea though. I suggest we explore providing an API for sleeping with regards to the simulation.

PeterJCLaw, 2020-06-12:
It might also be useful to distribute a linter configuration which bans the competitor code from importing the `time` module.

PR #110 (2020-06-12): Add simulation aware time and sleep methods
For reasons of practicality, these are part of the `Robot` object.

Resolves #101.

Built on #109 to placate mypy.
closed 2020-06-12 by PeterJCLaw