srobo/competition-simulator #114: Ensure Python processes are cleaned up when the simulation is reset

PeterJCLaw, 2020-06-12:
A report [on the forum](https://studentrobotics.org/forum/viewtopic.php?f=10&t=95) indicates that sometimes Webots fails to clean up the controller processes.

Possibly related, I've seen times when Webots notes that it's going to force-kill the controller because it didn't exit within one second. The docs imply that when Webots tells you (via return from the `step` method) that the simulation is going to end, that's a courtesy rather than a demand, yet the warning seems to disagree.

I'm not sure if these two are related, however it seems plausible that Webots is actually expecting the controller processes to quit ASAP after it informs them the simulation is going to end (rather than waiting to be killed).

antoinepetty, 2020-06-14:
This is quite noticeable when running multiple robots. The processes add up quickly

RealOrangeOne, 2020-06-15:
Looking briefly into how this works:

- Signals aren't sent to the `sr_controller.py` process before hard terminate
- Signals aren't sent to the `robot.py` process before hard terminate

Here's how the hard terminate works: https://github.com/cyberbotics/webots/blob/56bab11b4491c03fd6e9dee8fc64f2ee60b6a701/src/webots/control/WbController.cpp#L157

Also interestingly, the response from `webot.step` doesn't appear to change when the controller is told to stop. This might be because of the non-standard way we're running `step` in a background thread.

PeterJCLaw, 2020-06-15:
I wonder if we should `exec` rather than `fork` the `robot.py` process?

PeterJCLaw, 2020-07-03:
An alternative approach than #143 might be to inline execute the competitors' python code within the main controller process.

I'm fairly sure we could do that relatively easily. It would involve:
- setting the working directory to the new value (we already have this to hand anyway)
- reading in their `robot.py`
- calling `exec` on that code (passing empty state)

I would hope that this would avoid the proliferation of child processes and may even help clarify error messages from the child process as it would remove the indirection.

closed 2020-07-03 by PeterJCLaw