srobo/competition-simulator #116: Tokens fly around too easily

PeterJCLaw, 2020-06-12:
As reported in a [couple](https://studentrobotics.org/forum/viewtopic.php?f=9&t=87#p337) of [places](https://studentrobotics.org/forum/viewtopic.php?f=13&t=97&p=371#p371) in the forums, including a [youtube video](https://www.youtube.com/watch?v=gZ1hfbmmyX4).

This potentially means that the tokens will become somewhat challenging just to move around, as well as making it possible for them to get thrown out of the arena.

RealOrangeOne, 2020-06-12:
This would likely be fixed by some modified version of https://github.com/srobo/competition-simulator/pull/104

PhilR54, 2020-06-12:
Token mass is currently set far too low at 1g - will implement a fix alongside #100, making it easier to adjust the values in the future.

closed 2020-06-14 by PhilR54