srobo/competition-simulator #118: Robot has no attribute usbkey

PeterJCLaw, 2020-06-13:
Not sure if this is an intentional difference from the SR API, though it would be an easy one to fix.

Some competitors use this to know where on the filesystem their config files (or similar) are.

RealOrangeOne, 2020-06-13:
This likely won't play nice with #102, as the users code is mounted read-only.

PeterJCLaw, 2020-06-13:
I'm not sure I understand why there's a conflict there?
If we need to document that the filesystem is read-only that's fine.

Does Firejail move the code at all, or does it keep the same apparent path? I suspect without this competitors will resort to hard-coding things which are unlikely to work.

RealOrangeOne, 2020-06-13:
Firejail keeps the same code, so we're fine on that front. Documenting that the filesystem may be read-only in a competition environment would solve that issue, though.

closed 2020-06-13 by RealOrangeOne