srobo/competition-simulator #130: Robots can see tokens which fall outside of the arena

antoinepetty, 2020-06-14:
If a token gets moved outside of the arena by dodgy physics, the robots can still see it and may attempt to use it. Perhaps we need to raise the wall

PhilR54, 2020-06-14:
Could a supervisor be used to keep track of the tokens and change their properties when they leave the arena?

PeterJCLaw, 2020-06-14:
What's the core issue here?

Is it that the tokens can be "seen" _through_ the walls of the arena? Or that they can be seen over the walls?

The former feels like a substantial issue in how our vision works -- presumably it also means that tokens can be seen through robots, other tokens and the central pedstal too?
If so, then we need a more general fix for the vision system.

The latter feels like it's a non-issue and exactly what might happen in a real competition if there were a token that was thrown out of the arena.
If so, I think this is a non-issue and we should concentrate on whether we want to allow tokens to leave the arena in the first place.

antoinepetty, 2020-06-14:
They're being seen over the wall, not through it

PeterJCLaw, 2020-11-21:
Blocked: for SR2021 we're not running with a camera, so this isn't relevant this year.
I'm leaving this open however as the camera code is still present and still has this bug. If we go back to having the camera we'll need to fix this.