srobo/competition-simulator #176: Reload world causing seg fault when SWI code loaded

JennyLynnFletcher, 2020-07-11:


RealOrangeOne, 2020-07-11:
This seems to be an issue in collaboration with something in the supervisor. Perhaps around it removing robots.

JennyLynnFletcher, 2020-07-11:
Looks like it's caused by record_animation() in competition_supervisor.py

PeterJCLaw, 2020-11-21:
@JennyLynnFletcher any idea if this is still an issue?
I'm suspecting it may not be because we're on a new version of the simulator now and we've changed how the supervisor works. Also we're not reloading the world anywhere.

PeterJCLaw, 2020-12-21:
I'm going to close this as we've not seen anything like this recently and reloading the world isn't something we expect to do in normal operations at the competition.
It _may_ still be an issue, but without a reproduce it's hard to know and if it's not affecting us this doesn't seem worth keeping open. We can re-open this with more details if it reappears.

closed 2020-12-21 by PeterJCLaw