srobo/competition-simulator #177: Cancelling video creation may cause seg fault

JennyLynnFletcher, 2020-07-11:
Not fully tested yet but cancelling instead of saving a video seems like it may cause seg fault

PeterJCLaw, 2020-10-24:
We think that upstream fixes may have helped with this. We should re-test it.

PeterJCLaw, 2020-10-25:
I've picked up #181 and gotten the automated recording basically sorted (it's pending #205, but that just needs review). Once that's done each match becomes a single command and we always run a fresh Webots for every match.
I've tested that behaviour and had it successfully output a video file for a match.

I'm therefore unsure whether this is relevant any more?
(If we have steps to reproduce then we should report upstream though)

JennyLynnFletcher, 2020-10-29:
https://github.com/cyberbotics/webots/issues/1956 contains more info. The behaviour I identified as causing the segfault seems to now work fine. Worth noting that reloading world in the middle of video recording will not stop the video recording and the reloading will be visible in the output file.

closed 2020-10-29 by JennyLynnFletcher