srobo/competition-simulator #199: Compatibility with Kit API

trickeydan, 2020-10-24:
This simulator currently exposes a Python API that is not compatible with our existing Python 2 kit, nor the upcoming [sr-robot3](https://github.com/srobo/sr-robot3).

Whilst the new sr-robot3 API is now delayed until SR2022 due to SR2021 being virtual, this is an issue that needs to be considered. I recall [pre-empting this issue](https://github.com/srobo/competition-simulator/issues/5#issuecomment-626163844) when this simulator was designed, although this was [subsequently ignored](https://github.com/srobo/competition-simulator/pull/32#issuecomment-628200440) in a PR that was merged within **3 minutes** of creation (no chance for review!).

Causing breaking changes to the API too frequently will confuse both students and volunteers, which is why we planned to originally limit the breaking changes to coinciding with the move to Python 3 and sr-robot3.

There is no ideal solution to this situation, and we can only learn from mistakes that have been made here. I suspect that the best solution to this is to make use of the functionality for multiple backends that is built into sr-robot3, and ship that as the API for SR2021.

trickeydan, 2020-10-24:
It looks like it shouldn't be too difficult to get this working with sr-robot3.

## Requirements

- Add sr-robot3 as a dependency of the competition simulator
- Subclass sr.robot3.Robot to add webots specific functions
- Implement simulator backends (see [https://github.com/srobo/competition-simulator/pull/32#issuecomment-628200440](https://github.com/srobo/competition-simulator/pull/32#issuecomment-628200440)
    - SRV4PowerBoardSimulatedBackend
    - SRV4MotorBoardSimulatedBackend
    - SRV4ServoBoardSimulatedBackend
    - SRV4CameraSimulatedBackend
    - SRV4RuggeduinoSimulatedBackend

Some additional work will likely be required to get vision working, but all of the hard work has been done for it already.

PeterJCLaw, 2020-10-24:
Are there substantial ways that this API breaks from the existing (Python 2) kit API other than due to missing/added features or by merely being Python 3 rather than Python 2? If there are concrete things which can be fixed then we can consider fixing that for SR2021.

We should also bear in mind that we now have a non-trivial number of volunteers who have experience with the API which this simulator exposes.

As an aside: we are also discussing whether or not we need to provide an API wrapper at all given that Webots' API is already in Python and that encouraging the teams to use it would allow them much more flexibility in the robot chassis that they enter.

trickeydan, 2020-10-24:
Note: I disagree with the classification of "nice to have", and believe this lives under "should have".

trickeydan, 2020-10-24:
> Are there substantial ways that this API breaks from the existing (Python 2) kit API other than due to missing/added features or by merely being Python 3 rather than Python 2?

I do not have a conclusive list, but there are definitely missing features, such as `use_brake`.

> We should also bear in mind that we now have a non-trivial number of volunteers who have experience with the API which this simulator exposes.

Unfortunately these volunteers were all taught a "temporary API", that will be replaced and was known at time of shipping that it would be replaced.

> As an aside: we are also discussing whether or not we need to provide an API wrapper at all given that Webots' API is already in Python and that encouraging the teams to use it would allow them much more flexibility in the robot chassis that they enter.

I would support this entirely, as it resolves this issue.

PeterJCLaw, 2020-10-24:
> > Are there substantial ways that this API breaks from the existing (Python 2) kit API other than due to missing/added features or by merely being Python 3 rather than Python 2?
> 
> I do not have a conclusive list, but there are definitely missing features, such as `use_brake`.

`use_brake` is a know issue: #93.

Unless there are other variations from the published SR API I suggest that we close this or at least drop it from the SR2021 milestone. We have very limited time in which to prepare the simulator for SR2021 and changes to a known-working API aren't likely to be a priority.

> > We should also bear in mind that we now have a non-trivial number of volunteers who have experience with the API which this simulator exposes.
> 
> Unfortunately these volunteers were all taught a "temporary API", that will be replaced and was known at time of shipping that it would be replaced.

Feels like you're supporting two different sides of the same thing here -- either we care about volunteers not needing to learn a new API or we don't. If we don't care, then it doesn't matter whether or not the API changes. If we do care then _any_ change (whether wholesale replacement or otherwise) needs consideration. Furthermore, changes to APIs which have actually shipped clearly have a much larger impact on mentors than comparisons to potential future APIs.

trickeydan, 2020-10-24:
Dropping from milestone until this issue inevitably pops its head up again.

For future reference, this issue and the related are not valid factors for suppression of the changes that sr-robot3 introduces, as I have been **repeatedly** warning that we need to be careful.