srobo/competition-simulator #200: Fully support wait_start

PeterJCLaw, 2020-10-25:
I think we're going to need to have a proper mechanism to have the robots wait for the start of the match if we want to have automatic video collection (#181) due to limitations in webots (https://github.com/cyberbotics/webots/issues/2386).

Since we previously ran the competition on Windows I'm assuming that we need to support doing this on Windows too. Obviously we'd really like that anyway in order to support development in competition mode on Windows.

The way that this works on the real robots is that a FIFO is created using `os.mkfifo`, however that's a Unix-only API so that's out.

The other alternatives I can think of are:
- use a `socket.AF_UNIX` socket
  Impractical as this isn't supported yet in Python (https://bugs.python.org/issue33408). It looks like it might land some time during Python 3.9, but it's not here yet.
- use a named pipe
  [pywin32](https://pypi.org/project/pywin32/) seems to be the consensus for using these on Windows, however that means adding a potentially complicated (and undocumented) dependency along with factoring out the backends to be platform specific
- watch a directory for a file appearing (or disappearing)
  This is horrible.

Any other ideas?

trickeydan, 2020-10-25:
In an ideal world, I'd suggest using `SIGUSR1` however that doesn't work on Windows :/

Easiest solution might be to pick a high TCP port that is not in the [IANA list](https://www.iana.org/assignments/service-names-port-numbers/service-names-port-numbers.xhtml) and wait for a connection on it. e.g `47623`

PeterJCLaw, 2020-10-25:
`SIGUSR1` would require at least knowing the PIDs of the robot controller processes though?

Yeah, a local TCP socket might be the way to go.

I'm also exploring whether there's something internal to Webots we can use.

PeterJCLaw, 2020-10-25:
> I'm also exploring whether there's something internal to Webots we can use.

[`wb_robot_get_custom_data`](https://www.cyberbotics.com/doc/reference/robot#wb_robot_get_custom_data) is looking quite promising for this. It allows a (webots) `Robot` to get or set a string about itself and is noted as being a way to communicate between a controller and robots.

PR #205 (2020-10-31): Implement wait_start via Webots customData field
This is closer to how we really run matches and removes the need to reset the simulation as part of the setup process.

In turn, this will make it possible to automate video capture of the matches without running into Webots limitations.

For the moment I've left the animation capture as starting at the start of the match time, however it could now start earlier (i.e: including the setup phase too) if we wanted.

Suggest review by commit as I needed to fix some type annotations first.
~Note: I've based this on `improve-dev-ux` (#204) for my ease [... cut]
closed 2020-10-31 by PeterJCLaw