srobo/competition-simulator #279: Document Mac setup better

PeterJCLaw, 2021-01-13:
We know that you need to set the path explicitly -- we should call this out explicitly.

PeterJCLaw, 2021-01-13:
For `pyenv` it looks like you need to do this:
``` shell
PYTHON_CONFIGURE_OPTS="--enable-framework"  pyenv install --force  3.7-dev
sudo ln -s  ~/.pyenv/versions/3.7-dev/Python.framework /Library/Frameworks/Python.framework
```

PeterJCLaw, 2021-01-13:
The path to explicitly set is then:
`/Library/Frameworks/Python.framework/Versions/3.7/bin/python3`

WillB97, 2021-01-30:
We should probably list common locations for python on all OSes in the installation docs.

PeterJCLaw, 2021-01-30:
I'm not sure that's needed? In linux you don't need to do anything special for it to work. You maybe need to change from `python` to `python3`, but even that may not be needed on some distros. If we think that Windows setup involves dancing I suggest we consider documenting that separately.