srobo/competition-simulator #282: Supervisor log not saved with run-comp-match

WillB97, 2021-01-14:
Since `run-comp-match` uses a temporary directory and copies out the relevant files after the match, any files that are not copied are lost.

This is especially problematic since uncaught stack traces are not always printed to console before webots exits.

PeterJCLaw, 2021-01-14:
This should already be handled -- in the case that the supervisor errors it will cause webots to exit with an error code. When that happens `run-comp-match` will print the log to the console itself before the temporary directory is cleared.
Do you have a reproducible scenario where this doesn't happen?

WillB97, 2021-01-14:
It's very intermittent but it's most common with the error for required libraries.

WillB97, 2021-01-14:
Failing to print the errors is rather intermittent but the error for required libraries causes this most often.