srobo/competition-simulator #290: Add flag to inhibit movie creation in comp mode

WillB97, 2021-01-22:
Since we are looking at testing code by running short matches in CI and the videos created are discarded, it is unnecessary to record the movie in the first place.

Not doing a recording greatly increases the speed of the simulation and as such decreases the CI time. On an example CPU-only system a 5-second match is reduced from 2 minutes 18 seconds to 24 seconds.

This would most benefit #271 and PeterJCLaw/code-submitter#10

PeterJCLaw, 2021-01-22:
Since the quality of the recording is (IIRC) already controllable via the match file, maybe an option is to have quality 0 imply this?

WillB97, 2021-01-22:
I was thinking simply having an environment variable like `SR_COMP_NO_RECORD` but using the match file also acheives a similar end.

PR #292 (2021-01-30): Add flag to inhibit movie creation in comp mode
Since we are looking at testing code by running short matches in CI and the videos created are discarded, it is unnecessary to record the movie in the first place.

Not doing a recording greatly increases the speed of the simulation and as such decreases the CI time.

Fixes #290
closed 2021-01-30 by WillB97