srobo/competition-simulator #305: Make Robot.time() count from the start of the match

WillB97, 2021-02-05:


PeterJCLaw, 2021-02-05:
This difference is expected and normal. It's also explicitly documented that the zero time here is the start of the simulation, so this would be a breaking change.

I am strongly against this, for a number of reasons:
- this would not have a well-defined meaning when in development mode
- this is not how `time.time` works on the Robot kits (and it would be complicated to make it behave like this on the kits)
- this is not representative of how general APIs work -- you don't usually find that APIs provide a value for your local convenience, rather they provide tools for you to work with

RealOrangeOne, 2021-02-05:
This is also super easy to custom implement:

```python
R = Robot()
start_time = R.time()
```

Should team need this, I think that's totally fine to require them to do, especially given it's explicitly documented and super simple.

WillB97, 2021-02-05:
How obvious to teams is it that time has slightly different properties in comp mode? If it's not obvious that there is this waiting period at the start of matches teams could quite resonably write something that works fine in dev but doesn't account for this delay.

PeterJCLaw, 2021-02-05:
> How obvious to teams is it that time has slightly different properties in comp mode?

I'm not sure that's a true framing of this. The behaviour of `Robot.time()` is documented as being in reference to the start of the simulation. The docs themselves already show that time should be measured as that between two measurements, rather than as an absolute value.

> If it's not obvious that there is this waiting period at the start of matches teams could quite resonably write something that works fine in dev but doesn't account for this delay.

That competition mode has different properties is well established (although perhaps not well documented). If this is proving to be an issue the we should fix _that_ issue.

Another option here might be to record the real `time.time()` value at the start of the simulation and add that to the value returned by `Robot.time()`. That would make it really obvious that the absolute value should not be used, without creating cases which are complex to solve (such as what to do outside competition mode). This would of course be a breaking change, so not something we can do this year.

WillB97, 2021-02-05:
From the discussion on discord and a look through the docs, the fundamental issue here is that at no point do we state that in comp mode the robot will wait at the `R = Robot()` line until the match begins. As such it is unclear to teams where to measure the start time from, especially as in dev mode the value returned both before and after that line are the same.

PeterJCLaw, 2021-02-05:
Ah, yeah. This was trivially obvious on the kits because of the start button. The simulator does somewhat mask that.

It would be easy enough to force a timestep during `wait_start` even in dev mode.

WillB97, 2021-02-05:
Ultimately this is probably a docs issue but adding a little delay in `wait_start` dev mode is probably a good move.

closed 2021-02-06 by PeterJCLaw