srobo/competition-simulator #351: Clicking 'Reload World' does not restart the robot

Adimote, 2022-10-21:
if you click the 'Reload World' icon:
[image]

It restarts the simulation, but it doesn't run the robot code

The log messages after a reload look like this:

```
INFO: sr_controller: Starting controller: python.exe -u sr_controller.py 0
INFO: sr_controller: Starting controller: python.exe -u sr_controller.py 1
INFO: sr_controller: Starting controller: python.exe -u sr_controller.py 2
INFO: sr_controller: Starting controller: python.exe -u sr_controller.py 3
INFO: competition_supervisor: Starting controller: python.exe -u competition_supervisor.py
INFO: lighting_controller: Starting controller: python.exe -u lighting_controller.py
No robot controller found for zone 1
INFO: 'sr_controller' controller exited successfully.
No robot controller found for zone 2
INFO: 'sr_controller' controller exited successfully.
No robot controller found for zone 3
INFO: 'sr_controller' controller exited successfully.
Development mode, exiting competition supervisor
INFO: 'competition_supervisor' controller exited successfully.
```

PeterJCLaw, 2022-10-21:
It looks like you've ended up paused, were you paused when you clicked restart? For me if the simulation is running then when I click restart the simulation starts immediately.

I'm also not sure this is something we've done -- I think you're just experiencing webots behaviour.

PeterJCLaw, 2022-10-23:
We dug into this a bit yesterday. It turns out to be to do with the lighting controller constructing a `Supervisor` (robot) instance after the restart. That call never returned meanwhile the user-code doing a (Webots) `Robot.step` also never returned. Together this seemed to deadlock the simulation.
This suggests that this is a bug in Webots, however the latest nightlies of Webots don't have this issue so it may be fixed already.

PeterJCLaw, 2022-10-27:
For completeness: it looks like this was probably https://github.com/cyberbotics/webots/issues/5279.

PeterJCLaw, 2022-10-27:
I'm actually not completely certain that that situation aligns with what we saw, given that in our case there wasn't a crash, however the symptoms are otherwise very similar and the fact that the issue we observed seems fixed in the nightlies does suggest that it's been fixed around the same time.

closed 2022-10-27 by PeterJCLaw