srobo/competition-simulator #370: Tokens stick to robots

WillB97, 2023-01-25:
A team has reported that the tokens are sticking to the robot.
https://discord.com/channels/1026134973205463070/1026212782477148211/1067486566584426516

Adimote, 2023-02-05:
I can reproduce this quite often (about 1 in 3 times) with the gold token:

https://user-images.githubusercontent.com/2551763/216829133-67075885-1637-48f4-9207-0dcdb885a32a.mp4

Once it's stuck its really hard to unstick, so I'd mark this pretty high priority.

Likely this is caused by the markers on the tokens having weird collisions. Fix is likely to disable collisions for the faces

PeterJCLaw, 2023-02-05:
> Fix is likely to disable collisions for the faces

Presumably you mean the markers?

Yeah, the markers are separate entities which are just slightly sticking out from the tokens. If we can turn off their physics while still having them being recognition objects that sounds like a good fix.

Adimote, 2023-02-05:
Confirmed that the markers are the problem, removing them makes the physics a lot more robust:

https://user-images.githubusercontent.com/2551763/216830204-e7f8a6bd-bc55-4f42-b1b9-f5663c45d82b.mp4

I tried disabling collisions by setting `enableBoundingObject FALSE` in the marker solids but it didn't seem to have an effect.

Adimote, 2023-02-05:
I've now also tried just setting `physics NULL` for the in `TokenMarker.proto`, even though there's a comment saying it works fine with it set to null, it actually doesnt. The cubes seem to freeze in space as if the markers aren't allowed to move.

Adimote, 2023-02-18:
One thing to try tomorrow:
Someone mentioned that we could clear the bounding boxes for the markers, but keep the boxes just for their appearance

Adimote, 2023-02-18:
https://user-images.githubusercontent.com/2551763/219881901-5f2da4ef-4e0e-41b5-8ff3-6073f7c65608.mp4

some very weird behavior if I don't define the bounding boxes for the markers

Adimote, 2023-02-18:
If I remove both physics and bounding boxes, the boxes behave correctly!

However the camera detections seem way off.

With bounding boxes & physics:
[image]

Without bounding boxes & physics:
[image]

PeterJCLaw, 2023-02-24:
We have not managed to fix this.

closed 2023-02-24 by PeterJCLaw