srobo/competition-simulator #378: run-comp-match animations are broken

WillB97, 2023-02-13:
Webots animation files now record the absolute path to their associated files so copying them from a temp directory causes them to not load correctly.

PeterJCLaw, 2023-02-13:
Hrm, this seems non-ideal. A simple fix might be to find/replace the directory in the generated files. We should also report upstream that using the absolute path makes them non-portable.

WillB97, 2023-02-18:
This looks to not effect the MacOS version so may be a windows specific issue.

PeterJCLaw, 2023-02-18:
Looking at a recent animation I've got on Ubuntu the issue doesn't seem present here either (the paths are all relative).

PeterJCLaw, 2023-10-31:
Reported upstream as https://github.com/cyberbotics/webots/issues/6426.