srobo/competition-simulator #381: `sr.robot` not found if another `sr.something` namespace package installed in same Python

PeterJCLaw, 2023-02-17:
If using the same Python environment to run e.g: SRComp (which installs to `sr.comp`) as the one you want to use for running the simulator then the `sr.robot3` package will not be found even though it is correctly on `PYTHONPATH`.

The fix seems to be to change `modules/sr/__init__.py` to `modules/sr/__init__.pyi`. This changes the package being an implicit namespace package, which is compatible with SRComp at least.

trickeydan, 2023-02-17:
https://peps.python.org/pep-0420/

PeterJCLaw, 2023-02-17:
Indeed. The `.pyi` file is useful to ensure that `mypy` correctly handles the namespace packages (or it certainly used to be needed, that may have been fixed).

trickeydan, 2023-02-18:
> Indeed. The `.pyi` file is useful to ensure that `mypy` correctly handles the namespace packages (or it certainly used to be needed, that may have been fixed).

I think there's a --namespace-packages arg for mypy to handle that. We use it for sr.robot3

closed 2023-09-09 by PeterJCLaw