srobo/competition-simulator #408: Improve setup docs

PeterJCLaw, 2023-09-17:
In SR2023 several teams apparently struggled to set up the simulator for use. If we can make the process easier that would be great, though likely the best we can do (for SR2024) is document it better.

PeterJCLaw, 2023-09-17:
@8BitJosh as this suggestion came from you - do you have any details of specific issues which competitors hit in SR2023 which we should address?

8BitJosh, 2023-09-23:
Should go through the full setup process and use images/videos to explain the process.
We should look at any steps that may be confusing to improve how it is setup.

Also I had feedback from competritors that they found it difficult to undestand the text based file tree.